dekuNukem/exixe #1: LED does not stay lit

modalexii, 2018-03-24:
I think when you zero out the buffer in `exixe::set_led` and then `spi_write()` out that buffer, the LED gets turned off. Maybe unrelated, the LED does not light at all when `set_led()` is called in arduino's `setup()`.

dekuNukem, 2018-03-24:
Did LED work at all?  What kind of Arduino are you using?

I tested the library with an Arduino Uno and Arduino Micro and it seemed fine. I'll try it again tomorrow.

If there is no response from the module at all, then it's more likely something wrong with the wiring, maybe tell me more about it and I'll try to figure it out.

dekuNukem, 2018-03-24:
Here is a barebone sketch that doesn't use the library at all, maybe try it and see?

https://github.com/dekuNukem/exixe/blob/master/arduino_examples/barebone/0_LED_test/0_LED_test.ino

modalexii, 2018-03-25:
The LED works. Using Teensy 3.2 (& 3.5). The barebone sketch works, but it's different than the library - in both barebone sketches, the buffer is never cleared like it is in line 67 of `exixe.cpp`

modalexii, 2018-03-25:
Superficially this seems to fix it (I would submit a pull but the code is locked up in that zip :-) ):

`diff exixe.orig.cpp exixe.new.cpp `

```diff
67c67
<   memset(spi_buf, 0, EXIXE_SPI_BUF_SIZE); // clear SPI buffer
---
>   memset(spi_buf, 0, EXIXE_SPI_BUF_SIZE - 3); // clear SPI buffer except for LED 
83d82
<   memset(spi_buf, 0, EXIXE_SPI_BUF_SIZE);// clear SPI buffer
136c135
< }
---
> }
```

dekuNukem, 2018-03-25:
well thanks modalexii! I'll update the library tomorrow and test it again.

dekuNukem, 2018-03-26:
I have updated the library file, please check it out and see if the issue is fixed

I also added the library files outside the zip file, so feel free to submit pull requests.

closed 2018-04-02 by dekuNukem