dekuNukem/exixe #2: Arduino library: Initialising multiple tubes doesn't seem to work

jbmorley, 2018-04-02:
I'm finding that initialising multiple exixe instances stops them all from working. For example,

```c
exixe_1 = exixe(cs_pin_1);
exixe_2 = exixe(cs_pin_2);
```

Individually each exixe works, but together, they do not.

I believe this is because the SPI initialisation is performed in the `exixe` method, and therefore called twice. This bit:

```c
  SPI.begin();
  SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
```

Removing this code, and performing it explicitly, just once, beforehand seems to work for me.

I propose introducing a new method, perhaps `begin`, or `setup`, which wraps the SPI setup and means that developers can still benefit from the convenience of using just the exixe library, but can also configure this all themselves if they want to.

dekuNukem, 2018-04-02:
I tested the multi module sketch on an Arduino Uno and Sparkfun Pro Micro and it was working, so I guess it's a ESP32 issue.

Still, it's a good practice to only initialising the SPI once. I'll rewrite and add a `exixe.begin` function tomorrow.

jbmorley, 2018-04-03:
Sounds great. Thanks!

closed 2018-05-07 by dekuNukem