dekuNukem/exixe #3: Arduino library: memset not defined

jbmorley, 2018-04-02:
I found that the library (used with the examples) wouldn't compile for me when targeting my Adafruit Huzzah32 ESP32-based board.

The specific issue appears to be that it can't find `memset`. I'm still not very familiar with Arduino and its idiosyncrasies, but some Googlging reveals that this can be fixed by importing Arduino.h.

I've added the following to the top of exixe.cpp and everything works fine:

```
#include "Arduino.h"
```

This could be an issue with my setup but it is, as far as I can tell, the default configuration for the ESP32 boards and I'm not aware I've done anything strange.

dekuNukem, 2018-04-02:
Thanks for the discovery, I don't have a ESP32 myself and apparently it's different.

Isn't `memset` in `string.h`?  Try`#include <string.h>` instead of `Arduino.h` and see if it works. I'll test and update the library tomorrow.

jbmorley, 2018-04-03:
A collection of other symbols seem to be missing with just `#include <string.h>`.

```
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp: In constructor 'exixe::exixe(int)':
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:18:19: error: 'OUTPUT' was not declared in this scope
   pinMode(cs_pin, OUTPUT);
                   ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:18:25: error: 'pinMode' was not declared in this scope
   pinMode(cs_pin, OUTPUT);
                         ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:19:24: error: 'HIGH' was not declared in this scope
   digitalWrite(cs_pin, HIGH);
                        ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:19:28: error: 'digitalWrite' was not declared in this scope
   digitalWrite(cs_pin, HIGH);
                            ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp: In member function 'void exixe::spi_write()':
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:25:24: error: 'LOW' was not declared in this scope
   digitalWrite(cs_pin, LOW);
                        ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:25:27: error: 'digitalWrite' was not declared in this scope
   digitalWrite(cs_pin, LOW);
                           ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:26:11: error: 'delay' was not declared in this scope
   delay(10);
           ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:29:24: error: 'HIGH' was not declared in this scope
   digitalWrite(cs_pin, HIGH);
                        ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp: In member function 'void exixe::crossfade_init(unsigned char, unsigned int, unsigned char, unsigned char)':
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:107:34: error: 'millis' was not declared in this scope
   animation_start_frame = millis() / EXIXE_ANIMATION_FRAME_DURATION_MS;
                                  ^
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp: In member function 'unsigned char exixe::crossfade_run()':
/Users/jbmorley/Documents/Arduino/libraries/exixe/exixe.cpp:116:41: error: 'millis' was not declared in this scope
   unsigned long current_frame = (millis() / EXIXE_ANIMATION_FRAME_DURATION_MS) - animation_start_frame;
                                         ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/jbmorley/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board Adafruit ESP32 Feather.
```

dekuNukem, 2018-04-03:
I updated the library just now so try it out. 

Feel free to submit pull requests too if there are any other issues.

jbmorley, 2018-04-14:
Thank you so much!

closed 2018-05-07 by dekuNukem