dekuNukem/exixe #6: exixe12_bom different capacitor values

BenRoe, 2018-11-15:
Hi,
in the [exixe12_bom.xlsx](https://github.com/dekuNukem/exixe/blob/master/resources/exixe12_bom.xlsx)
C1 has a capacity of 1uF, but in the eagle file it has 100nF (0,1uF).

100nF is the correct value?!

Btw thanks for sharing.

dekuNukem, 2018-11-15:
The capacitors on this board are for decoupling purposes only, so their values doesn't matter that much. Anywhere between 100nF and 1uF works, I used 100nF for my production run and hasn't had any problems.

BenRoe, 2018-11-15:
Ah ok. One more question. Is 0,125 W enough for the resistors?

dekuNukem, 2018-11-15:
Yes, I've measured that the voltage drop at the anode resistor is 44V, that with 68KΩ consumes 0.0284W of power, well below 0.125W.

closed 2018-11-15 by BenRoe