dekuNukem/exixe #11: Exixe14 BOM not matching PCB

eben80, 2020-02-22:
Hi,
The labels on the pcb goes up to R28 while the BOM only goes to R25.
This means that i don't know the values for the bottom of pcb. 

Thank you

dekuNukem, 2020-02-22:
Is R28 on the bottom layer? If so it is not needed. No components are soldered on bottom side.

eben80, 2020-02-23:
Hi,
Yes the highest R value on the top layer is R24. 
So none of the 5 resistor pads need to be populated on the bottom? 
I followed the markings on the resistors of your top layer with the posted photo. 
Do you by any chance have a photo of the completed bottom of an Exixe14? 
Thank you for an awesome design.

dekuNukem, 2020-02-23:
There are no components on the bottom layer at all, let me see if i can find a photo later today.

eben80, 2020-02-23:
OK thank you.

dekuNukem, 2020-02-23:
Here's a photo of the bottom side:

[image]

eben80, 2020-02-23:
Great thank you,
So just to confirm : Even though there's pads for resistors on the bottom,
they should not be populated when assembling my own.

dekuNukem, 2020-02-23:
that is correct.

eben80, 2020-02-24:
Thank you very much.

closed 2020-02-24 by eben80