dekuNukem/exixe #15: Crossfade issue on ESP8266

eben80, 2020-03-31:
Hi,
I'm having issues getting the crossfade function to work correctly on my ESP8266 based device.
It keeps on going into software watchdog reset because of the while loop.
Adding yield(); in the while loop hasn't helped either.
Do you have any experience on running this successfully on these devices?

[image]

This is my loop code with the yield as required by ESP8266:
`void loop()
{
  count = (count + 1) % 10; // keep count between 0 to 9

  my_tube1.crossfade_init(count, 15, 127, 0);
  my_tube2.crossfade_init(10-count, 15, 127, 0);
  // my_tube3.crossfade_init(count, 15, 127, 0);
  // my_tube4.crossfade_init(10-count, 15, 127, 0);

  /*
      again, crossfade_run() is non-blocking, that means
    you can run other tasks in the loop.
      just make sure to call it at least every 33ms
    and check its return value to see if it's finished.
  */
  while(1)
  {
    yield();
    unsigned char result1 = my_tube1.crossfade_run();
    unsigned char result2 = my_tube2.crossfade_run();
    // unsigned char result3 = my_tube2.crossfade_run();
    // unsigned char result4 = my_tube2.crossfade_run();
    // exit when both animations are finished 
    if(result1 == EXIXE_ANIMATION_FINISHED && result2 == EXIXE_ANIMATION_FINISHED)
      break;
  }

  my_tube1.set_led(127, 40, 0); // orange;
  my_tube2.set_led(127, 0, 127); // purple;
  my_tube3.set_led(127, 40, 0); // orange;
  my_tube4.set_led(127, 0, 127); // purple;
  
  delay(500);
}`

dekuNukem, 2020-04-02:
Unfortunately I don't have experience with ESP8266 modules. However, by the sound of it, if it goes into watchdog reset, it might have been getting stuck somewhere in the code. Maybe it would help if you can find exactly which function is getting stuck, and see what's going on inside. The exixe library itself is very simple and straightforward too, so it probably won't be too hard.

eben80, 2020-04-03:
Thank you I will see if I can figure it out.

closed 2020-04-03 by eben80