dekuNukem/exixe #16: SPI pins

NegativCut, 2020-05-31:
Hi,are the green & blue addresses (spi_buf[14] and [15]) reversed in the Arduino 0_LED sketch? I find they are so I'm wondering if my boards have an issue. Tested 2x boards so far with the same results. exixe12. Thanks.

dekuNukem, 2020-05-31:
According to the [technical details](https://github.com/dekuNukem/exixe/blob/master/technical_details.md), spi_buf[13] should be red, spi_buf[14] should be green, and spi_buf[15] should be blue.

Are you saying on your boards spi_buf[14] and spi_buf[15] are reversed? As in spi_buf[14] is blue and spi_buf[15] is green? If true this is very bizarre indeed, I have never had this issue myself. 

Have you tried the [Arduino examples with library](https://github.com/dekuNukem/exixe/blob/master/arduino_examples/1_LED_test/1_LED_test.ino)? And which Arduino board are you using?

NegativCut, 2020-06-01:
Arduino Nano being used.

Here is the code used to get a purple LED... left as original it's yellow.

`/*
  exixe modules:
  https://github.com/dekuNukem/exixe

  library docs:
  https://github.com/dekuNukem/exixe/tree/master/arduino_library

  Demo 1: LED test
*/

#include "exixe.h"

// change this to the cs pin you're using
int cs_pin = 10;
exixe my_tube = exixe(cs_pin);

void setup()
{
  my_tube.spi_init();
}

void loop()
{
  my_tube.set_led(127, 127, 0); // purple
  delay(500);
}`

dekuNukem, 2020-06-01:
This is very strange indeed. This project is more than 3 years old and it is my first time hearing this issue. I'll do a little testing myself.

dekuNukem, 2020-06-02:
Turns out all my Arduinos are at my workplace, and its closed.

I purchased a cheap one online, but it will take a day or two to arrive.

In the meantime, if its convient to do so, can you take a look at your exixe12 module and tell me what it says at the location below? Thanks.

[image]

NegativCut, 2020-06-02:
Hi, the area shown on the boards I have is blank, If you need a picture, I can supply one later today.

dekuNukem, 2020-06-02:
that would be great, thanks.

NegativCut, 2020-06-02:
[image]

NegativCut, 2020-06-02:
This one I am building at the moment.

dekuNukem, 2020-06-02:
woah this is a very old revision of the circuit board. It looks like a prototype-stage design from 2017. I didn't know you're building one yourself, I thought you purchased one from my Tindie shop. 

Anyway, because this PCB revision is a very early design, I might have changed a few things around since then. The current library and documentation is for the production-run boards. If the swtiched RGB channel is the only issue, then thats actually not too bad!

NegativCut, 2020-06-02:
Yea I bought these off Tindie just during lockdown I've managed to get through a few projects :)
I have uploaded the hex from the repo and am using an Arduino, and all seems well except the RGB...

NegativCut, 2020-06-02:
[image]

Is this normal behavior running 2_loop_digit_simple?

dekuNukem, 2020-06-02:
It seems to be running a bit fast. In the [arduino code](https://github.com/dekuNukem/exixe/blob/master/arduino_examples/2_loop_digit_simple/2_loop_digit_simple.ino) the digit should be changing every half second, it looks about twice as fast in this gif.

The tube also turns off and flicker for a bit in the middle, if it happens on its own I would start by checking the 3.3V power supply. The 3.3V pin on 5V Arduino can hardly provide enough current for one module, so you might want to try a more beefy power source and see if it improves. I mentioned this issue in the [power consumption considerations](https://github.com/dekuNukem/exixe/blob/master/getting_started.md#power-consumption-considerations) sections of the user manual.

NegativCut, 2020-06-02:
Hmm, I'm running 5v through a 3v3 LD33 separate regulator, max 1A.

The other example, crossfade, is giving weird results. I'm going to try another Arduino to drive the spi.

dekuNukem, 2020-06-02:
The reset pin on the module is pulled up internally, so you don't have to pull it up again, but it wouldn't hurt.

I would check to make sure the SPI output from the arduino is at the correct speed and configuration with a logic analyser or osciliscope. You can [find the details here](https://github.com/dekuNukem/exixe/blob/master/technical_details.md).

NegativCut, 2020-06-02:
Ok, before I break out the saleae, I read the clock speed was tested up to 8Mhz, was there minimum viable during your tests?

dekuNukem, 2020-06-02:
Anything between 1MHz and 24MHz should work, although i would try slower speeds first to be on the safe side.

NegativCut, 2020-06-03:
Hi, so after connecting my Arduino and 3V3 grounds together the 2-digit counting and crossfading sketches are working just fine. I have tested the SPI clock up to 16Mhz and its working just fine.
Thanks for your help and feedback.

dekuNukem, 2020-06-03:
Yeah it's important that everything are at the same GND. Glad to hear it's working now.

closed 2020-06-03 by dekuNukem