dekuNukem/exixe #19: Random blinking numbers and inconsistent LED behaviour

Eternoxy, 2020-11-07:
Hello there,

i got myself some nixie tubes and the IN14 driver boards.
I tried hooking up the cables as shown in the getting started page and double checked the connections.
Then I tried running the LED-test code as prescribed. I could change the colour of the LED, but it also happened, that the LED changed randomly from bright to low white(ish) and to purple. So no consistent behaviour here.
When I tried the digit loop, the LED start blinking random, changing colour between purple and red and the digits of the Nixie tube show up, but in a random way and definetly not all 0.5s.
What could be possibly wrong and what else could I try to get this thing working?

I figured out several things what could be, but I´m not sure how to fix this.
What do you mean with RST must be high/low and how can I change this?

Should i attach a video with my current setup?

I´m not an electronic expert and I´m just starting to dive in the world of Arduino & Co, but I still want to figure out whats wrong and get this thing working. Unfortunately I dont own a logic analyzer yet. Would such a thing help figuring out the issue?

As a result i want 6 Nixie tubes showing a clock and changing to user defined values whenever i want it.
Its maybe still a long and bumpy road till there, but I think I can make it with some help.

dekuNukem, 2020-11-07:
Hi eternoxy! It would be great if you can provide a photo / video of your setup so I can take a closer look.

From what you describe it seems that the SPI command might be corrupted one way or the other. Does it happen to all the modules, or just one? The RST pin should be connected to 3.3V.

Eternoxy, 2020-11-07:
Hello dekuNukem,

thank you very much for your quick reply.
I did some more desperate testing and got some positive results with the RST connected to 3.3V but I´m still not there yet.
But at least I can say that the wiring should be correct.

I can now reliable change the LED colour. I can verify it by changing the RGB values in the arduino sketch. Sidenote here: I have to reconnect the power supply (sometimes several times) after uploading the sketch, the reset button won´t work to see an effect.

Next step would be getting the simple digit loop to work. 
But when i upload the sketch and reconnect the power supply, i get a blue LED and nothing else.
I can confirm that the nixie tubes and the power supply works, because when i disconnect the 3.3V from RST pin, "magical" things happen and the whole thing goes blinkie blinkie but not the way its supposed to ;)

Any ideas?
I can provide a photo or video tomorrow. The way its built now you probably wont be able to see any mistake between all the beginning mistakes im currently doing.

Eternoxy, 2020-11-08:
Ok, after some more testing i could also get the simple digit loop kind of to work.

This gives me a red 3 after uploading the code and reconnecting RST to 3.3V.

`  my_tube.set_led(127, 0, 0); 
   my_tube.show_digit(3, 127, 0);`

How will the digit loop work without reconnecting the RST pin every time?
Does it still need kind of a resistor?

dekuNukem, 2020-11-08:
I guess you can connect the RST pin to a digital pin on arduino. At the very start of the sketch make it go down like 50 milliseconds then back up, this will reset the module, then you program can start as usual :)

Eternoxy, 2020-11-08:
I´ve tried this now and it worked out:

`/*
  exixe modules:
  https://github.com/dekuNukem/exixe

  library docs:
  https://github.com/dekuNukem/exixe/tree/master/arduino_library

  Demo 2: loop digits
*/

#include "exixe.h"

// change this to the cs pin you're using
int cs_pin = 10;
unsigned char count;

exixe my_tube = exixe(cs_pin);
int rst_pin = 9;
void setup()
{
  my_tube.spi_init();
  my_tube.clear();
  pinMode(rst_pin, OUTPUT);
  digitalWrite(rst_pin, LOW);
  delay (50);
  digitalWrite(rst_pin, HIGH);
}

void loop()
{
  count++;
  my_tube.set_led(127, 0, 0); // purple
  /*
  1st arg: Digit to show, 0 to 9
  2nd arg: Digit brightness, 0 to 127
  3rd arg: Overdrive, 0 disable 1 enable
  */
  my_tube.show_digit(count, 127, 0);
  delay(500);
}`

Pin 9 is connected to RST.

I hope this  helps someone out. I can´t understand though why it worked for you without connecting the RST pin.

dekuNukem, 2020-11-08:
Great to hear that you got it working! I would put the reset right at the very beginning, and also give it a delay after it goes high. Something like this:

```
void setup()
{
  pinMode(rst_pin, OUTPUT);
  digitalWrite(rst_pin, LOW);
  delay (50);
  digitalWrite(rst_pin, HIGH);
  delay(50);
  my_tube.spi_init();
  my_tube.clear();
}
```
But if it works, it works!

The reason it worked for me without connecting to the RST pin is that it is weakly pulled up internally with a resistor around 13K, so it would stay high if nothing is connected to it. However if the circuit is noisy it's better to be sure and connect it properly.

Eternoxy, 2020-11-09:
Ok, i´ll try it also this way.
Thanks again for your quick and super helpful support!!!

I might have to ask something again, but thats it for now.
Wish me luck with the coding for the clock.

closed 2020-11-09 by Eternoxy