dekuNukem/exixe #20: The exixe 2018 boards not working

TheMobas0, 2021-08-18:
I have two exixe 2017 boards that work just fine, but when i tested the exixe 2018 boards with the same set-up, they didn't work.

dekuNukem, 2021-08-18:
hi there! Can you describe the issue in a bit more detail? Also let me know your order number and I can look it up too.

TheMobas0, 2021-08-18:
Hi, my order number is #273464. I've attached two nixies to two new 2018 exixe-14 boards, and then I put them on a breadboard and connected it, but then I turned it on the nixie tubes weren't working. I tried that exact set-up with my two other 2017 exixe boards and they worked just fine. Also am using a Raspberry Pi 4 and Python.

TheMobas0, 2021-08-19:
[image]
Added a photo to show one 2018 exixe board and one 2017 exixe board for context.

dekuNukem, 2021-08-19:
The soldering on the pin header on the upper board seems to be a bit iffy, especially the reset pin, I would suggest make sure they have good contact and no cold solder joints.

I would also suggest try starting with the simplest example, [LED test](https://github.com/dekuNukem/exixe/blob/master/arduino_examples/1_LED_test/1_LED_test.ino), see if the LED changes colour, and go from there.

Also, the year and production run is printed in the back of the modules, feel free to let me know the markings on yours.

[image]

TheMobas0, 2021-08-19:
LEDs work on both the 2017 and 2018 boards. My working 2017 boards have the number on the back but the 2018 don't. I attached pictures of both back and front of my 2018 boards.
[image]
[image]

dekuNukem, 2021-08-20:
You have tested the tubes before installing them right? That might well be the issue here. Make sure the HV pin is soldered properly.

Anyway, if it turns out to be an hardware issue with the board, I can send you a few replacement modules with the latest production date so you can try them out.

TheMobas0, 2021-08-20:
Yes, I tested both tubes before soldering them onto the boards. Also i rechecked the HV pin and its soldered properly.

dekuNukem, 2021-08-20:
Sure, I'll send you two replacement exixe14 boards to try out, same address as the order #273464?

TheMobas0, 2021-08-21:
Sorry, I made a mistake, it's actually order #270289 (the order where I purchesed the 2 not working exixes). Even Though the adresses should be the same.

TheMobas0, 2021-09-14:
Hello, I recieved my replacement exixes yesterday and finished desoldering and soldering today. The exixes work perfectly and the low-temprature solder was perfect for desoldering the nixies off the broken exixes. Thank you.

dekuNukem, 2021-09-14:
Glad to hear it worked out! Those low temp solders are amazingly versatile, perfect for desoldering chips without damaging them!

closed 2021-09-14 by dekuNukem