dekuNukem/exixe #22: Created and assembled board, connected to ST Link V2

Mmma2345, 2023-06-21:
Hi DekuNukem, 

so, firstly a massive thanks for the project; this is exactly what I’m looking for ! 

I’m also a beginner in electronics. But am a very quick learner. Before starting this I have done research and practiced using various boards etc. 
so I believe my soldering job is half decent. 

I’ve had the boards printed (Exixe-14)
I’ve assembled 1 board so far using the BOM provided and attached a working (and tested) IN-14 Nixie Tube. 

I’ve soldered pins to pads : 
3.3v
GND
DIO
CLK

from here I’ve attached the four pins via jumper wires to a ST LINK V2 as suggested. With the suggested software. 

I click the button to link to the STM32 via the software but an error appears showing : no st link found ( I’m writing this from memory so will double check the wording of this when next at the pc! )

I also checked the voltage across the GND and 3.3v pins when connected to the ST LINK V2 with a multimeter; this only showed around 0.1volts. 
but once disconnected the ST LINK V2 showed 3,3volts again. Seems odd ? 

Any helps would be great!!

Mmma2345, 2023-07-02:
Turns out I had on positions C2, C3 a Feed through Capacitor, which was reducing the voltage to the circuit?

So Ive replaced this for a normal/ correct one and now the circuit has power.

But now/ still I am having issues connecting to the STlink Utility, it does not connect.

Please, if there is anything you can suggest to do?

Michail1, 2024-04-04:
If you include pictures, it may help in determining your problem.  After the CAP changes, were you able to have 3.3v?  Provide a HD picture of your board and I might be able to spot any issue.  (I've made many board and they work).  Also, provide picture of your STLink wiring.  Might be something as simple as wrong wiring.
Just try a firmware up on the STLink without anything connected just to see if the computer can at least talk to the programmer.
Then wire and try the board or boards you have made.  eMail me directly for other help and pictures.