dekuNukem/exixe #23: STM32 ST-LINK Utility Error (Can not connect to target!)

ccmilbrot, 2024-05-26:
I'm new to building and SMD components to pcb and have run into a problem trying to load the firmware to the STM-32  module. Attached is a photo of the error I'm running into, plus the front and back of the Exixe-12 board I soldered the components onto (maybe I messed that up).

Hoping someone can help me out or point me in the right direction.

[image]
[image]
[image]

ccmilbrot, 2024-06-02:
I solved this on my own. The cables that came with my ST-LINK V2 clone were too long at 18cm. I cut all the wires down to 10cm and reconnected them between the ST-LINK V2 and the Exixe12 board and launched STM32 ST-LINK Utility and it immediately connected to the target
[image]
[image]

Michail1, 2025-05-16:
Glad you got it to work, but I don't think the length was the issue.  My leads are about two feet without a problem; however, I don't use bread boards.  I wonder if the connection is simply poor enough to have caused the issue.  Also consider using a test rig with pogo pins or the pogopin rig in the github.

[image]