dekuNukem/exixe #25: Still viable?

jet438, 2025-05-08:
Is this still a viable project to buy a premade board?

nobodydontknow, 2025-05-09:
I'm also interested in this. I submitted once to PCBWay for another project (BlueSCSI) but that one had CSV files for the BOM and pick and play info. I didn't see that in this project.

Michail1, 2025-05-15:
I think he stopped making them a couple of years ago.  I have since started making my own since I use them alot.
You're welcome to email me for help/info.     M1@Michail.com

Michail1, 2025-11-13:
@Fraizler I don't have any right now.  I will be making more shortly.  I did have them posted to ebay under Michail1

Sorry, I do not have a CPL file.  I've been manually making them as I need them with any extra posted to the bay.
I guess I should really take time and make a placement file and just order them premade by China.

Another note... I did make a programmer using an M5Stack so that I don't have to find the files and using a computer each time I make the boards.  Saves time.  I never did get the POGO pin programmer to work and didn't get a response from deku; I made several attempts.

nobodydontknow, 2026-05-10:
@Michail1 what would it take to get a BOM file and pick-and-place file? That way any of us could just order complete boards from jlcpcb.com.